ploopyco/classic-trackball #4: PS/2 Output

equalunique, 2019-11-01:
An outdated protocol is an odd thing to ask for, but I thought I'd try anyway. Is PS/2 output something that you've considered? Do you have any thoughts on level of effort? I'm imagining it would not be compatible with the current PCB.

I'm interested in connecting a Ploopy Mouse to a keyboard running QMK firmware, as described [here](https://beta.docs.qmk.fm/features/feature_ps2_mouse) and shown [here](https://github.com/qmk/qmk_firmware/issues/5218#issuecomment-475496870).

Being able to connect it to QMK would allow me to bundle mouse and keyboard functionality into a single device, which paves the way to an integrated design, bluetooth, and more. :)

ploopyco, 2019-11-01:
Yeah, swapping the USB connector for PS/2 would be a new PCB, new hole in the enclosure, and new firmware. It'd basically be a new project.

[One of the QMK devs has been working on porting it over to this mouse](https://www.reddit.com/r/Trackballs/comments/dk4ha7/ploopy_vs_elecom_huge_now_with_qmk/). Just saying.

ChasonDeshotel, 2019-11-13:
That post is actually how I found Ploopy, love QMK, Ploopy Trackball looks amazing. Bluetooth would be sick

ploopyco, 2020-04-27:
Closing for lack of follow-up interest.

closed 2020-04-27 by ploopyco

mmuman, 2020-10-15:
Well, first generation USB mice had support for PS/2 using a passive adapter that just turned the USB plug into a mini-DIN.

It should just be a matter of detecting levels on the data signals and bit-bang PS/2. But sadly the chosen uC puts USB signals on single-function pins, so you can't reprogram them to be GPIOs. It may be possible to work around this if the USB core can be put to standby (and have D+/- as high-impedance, and if the current design has 2 free GPIOs. In this case, adding 2 wires should be the only mod required. The firmware would start by turning USB down, setting the GPIOs to detect the levels, set them up for PS/2 if detected, or turn them off and USB back on in the other case.