ploopyco/classic-trackball #5: Second Sensor and Twist Scrolling

kamegami13, 2019-11-01:
I'm curious if it's possible to add a second sensor so it can do twist scrolling like the Kensington Slimblade. On a similar note, if it was possible to use for 1:1 rotation control for CAD/modeling programs. My understanding is that with two sensors you could track all three axis' of rotation.

ploopyco, 2019-11-01:
At the minimum, my gut feeling is that it would require a third PCB (to house the sensor, which needs to be positioned from the ball at a very precise distance). Changes to the enclosure, too, to hold the new PCB, as well as a new hole.

I'm not thrilled about the idea. I wanted a very specific set of features in this design, and I'm happy with where it's ended.

ChasonDeshotel, 2019-11-13:
Sounds awesome. With [QMK running on this unit now](https://www.reddit.com/r/Trackballs/comments/dk4ha7/ploopy_vs_elecom_huge_now_with_qmk/f4akfnh/) writing custom code is possible, so I imagine the difficulties will be finding an extra input on the PCB (or making a new one) and modifying the enclosure

>the sensor [...] needs to be positioned from the ball at a very precise distance

Is this a physical limitation of the sensor itself or could it be overcome via a configuration variable on QMK?

>I'm happy with where it's ended

As you should be, this project is seriously badass

>My understanding is that with two sensors you could track all three axis' of rotation.

That sounds about right

closed 2020-04-27 by ploopyco