ploopyco/classic-trackball #9: Missing dowel in Appendix D: Hardware

Lilolbet, 2019-11-28:
First off, this looks like a pretty neat project. Very interested in building a mouse and upgrading from my Logitech M570. I however don't see the what dowel is required for the bearings (Step 17 in 3. Enclosure assembly). 

Will look though all remaining files (if I find it, ill reply with the answer). Regardless, this information is still missing on both the Hardware and Assembly pages.

ploopyco, 2019-11-29:
The roller bearing dowel can be found [here](https://github.com/ploopyco/mouse/blob/master/hardware/Mechanicals/STLs/RollerBearingDowel.stl) as a STL, and [here](https://github.com/ploopyco/mouse/blob/master/hardware/Mechanicals/STEPs/RollerBearingDowel.stp) as a STEP.

closed 2019-11-29 by ploopyco

Lilolbet, 2019-11-29:
Well this is embarrassing. Looking at the picture I was sure it was a metal part. 

Thank you for clarifying.