ploopyco/classic-trackball #11: Bluetooth?

mseflek, 2019-12-09:
Hi,

Fantastic project. Any plans to make a bluetooth version or add on?

King-Icewind, 2019-12-09:
I (contributor) am currently working on rebuilding this in KiCad. Along with adding USB C and a few other minor changes, I intend to add battery charging and Bluetooth support (in a future version). I haven’t decided on if it will be an add on to the main board or a new board altogether.

ploopyco, 2019-12-10:
Nice @King-Icewind. I'm excited to see these changes. Out of curiosity, what sort of batteries are you planning on using? NIMH? Li-ion? Any plans for the housing, or will the batteries just be taped to the enclosure?

King-Icewind, 2019-12-10:
I've only ever worked with Li-ion in my projects, so I'll likely go that route. I can have 3.3v+ in a smaller package. I haven't drawn anything yet, however, I think a 3D printed battery holder that shares the rear two screw mounts of the base might work.

diabl0w, 2020-03-14:
pinging @King-Icewind any updates on this? I also saw your reddit thread

King-Icewind, 2020-03-14:
The board is all designed; I just have to order some made to test. 

I put the project on a temporary hold with the whole COVID19 thing. 5 boards from China was around $100, otherwise it was $300+. Looks like I could probably order them now though.

diabl0w, 2020-03-14:
> @diabl0w The board is all designed; I just have to order some made to test.
> 
> I put the project on a temporary hold with the whole COVID19 thing. 5 boards from China was around $100, otherwise it was $300+. Looks like I could probably order them now though.

yes its certainly a strange time, I hope everyone stays safe.

I look forward to checking out the new boards, I hope they end up working out. nothing better than having your design work first try 😅 thanks for advancing this project and ploopyco for the original project

closed 2020-04-27 by ploopyco