ploopyco/classic-trackball #12: Scrolling has finiky bug

nerd7473, 2019-12-10:
Scrolling makes content move up and down rapidly at certain points while scrolling. How would we implement a patch? Is there a patch yet?

nerd7473, 2019-12-10:
Bug seems to be fixed when you unplug the mouse, and plug it back in...

King-Icewind, 2019-12-10:
Did you hand solder your board?

nerd7473, 2019-12-10:
No, I didn't this was the hand assembled one from ploopyco

nerd7473, 2019-12-10:
Works like a dream after you unplug and plug it... But not sure why the bug exists in the first place.

King-Icewind, 2019-12-10:
I asked because it may be an issue with the IR emitter and/or phototransistors, or scroll wheel. There are spaces/holes in the scroll wheel: when the spaces break the IR beam it will tell the MCU there's scrolling happening. You can check your scroll wheel to make sure there is nothing in the tiny holes along with checking the vertical board for anything in the way of the IR emitter and transistors. If one of the IR emitters is slightly angled, it may cause an issue too.

This is a bit of a guess at what is wrong, but hopefully you are able to figure it out.

nerd7473, 2019-12-10:
As I said, after unplugging and plugging it in... I works fine, I do have another trackball on the system and that's a Logitech M570... Zorin 15 (Basically GNOME Shell rebranded) I have also had the issue on my Ubuntu laptop (running Unity 7.4). But after unplugging and plugging the bug goes away as long as I don't lock or reboot the system... Otherwise no issues... So it may be that GNOME-based shells have an issue with working with this mouse...

ploopyco, 2019-12-18:
I've been working on addressing this, and I think that I've found the root cause: the scroll wheel itself isn't correct. I've since updated the STEP and STL file for the wheel to have a different number of holes.

Send me an email at contact [at] ploopy {dot} co and I'll get a replacement to you, if you don't have access to a 3D printer. Once you've tested it, we can revisit this issue.

closed 2020-04-27 by ploopyco