ploopyco/classic-trackball #15: Missing stl of electronics assembly jig

Lilolbet, 2020-05-05:
I apologize in advance, since I assume I am at fault here.

For the life of me I can not find the .stl file for the electronics assembly jig. Is this available? I have checked out various commits and I can not find it anywhere.

The jig is the following one:

![Image](https://camo.githubusercontent.com/66248733af3e0b12f129085036c497171f856f5d/68747470733a2f2f6c68332e676f6f676c6575736572636f6e74656e742e636f6d2f754167785a72564a65772d6271496c7757417a705566366345505a68524f6c6f31343974623650754548484f6b3630572d784367684a59376774786b693162384e45575239664e383879387966474a59737042365647464a3970654933676d2d44715a786348367639724e3734416c4e7442386c313766764c2d6457465666524d312d7362465f53304f59744b68386669316e33313752626736695a707567636f38646b6d75646a6c495a7468457a76505667524253324b39715a6a467a2d51626d4b5f5941453261387a35674f794556527277586e384d72706543723544554d646e724853716b394b7233344876634876414543504e586650777a72687930314a75464546717350666d43536e755551585f59316b41342d516b57744d7736764b6f55456877344174746f4f454e3570774876673534585939307833633262784d794d37476c665144743678565f316e6e573633785334545535485762435f4c386a366c532d4e653033784771686658455f46444b6b6650536d493658356e45565f772d6e5f2d306b6a6f6b39545a75573678794647786f44626c6e68735957726a79526a794467355f494d752d774144744632714c4c54773343594c744e396c774962533236704835334b487154696d3249776966734a447938566f49474d4c655a2d6249326f4952426e5264575a462d4a4a5f2d4a6168716f577a6e5f36692d474638676259344a64653561714c3763517842367654432d3963424a584c66444d4a5f317148734b5147326f5f353641397856526d6f304a44786d48467853336b416332474a49685f4d6f48423134384d3466744b5762414855517358714d322d41676c3147327433696774646e52624672426f3467677461346f35612d4a66516d3668647336484e39355748786f5f4a4d49553931642d766b76746e4e384b444157726e54796b594169456b767a3241656b66572d2d7757636c764546757977387239704c5a38337570533246666b687a477365714d3244664b61625572424941477a564d4849324e646f3d77313437322d683938312d6e6f)

ploopyco, 2020-05-05:
You're correct - it's not in the repository. They're only included in kits bought from the store.

Strictly speaking, that jig isn't necessary to complete assembly (as can be seen [here](https://github.com/ploopyco/mouse/wiki/2.-Electronics-assembly)).

closed 2020-05-05 by ploopyco

Lilolbet, 2020-05-05:
Ahhh! Thank you, I thought I was going crazy :)