ploopyco/classic-trackball #29: REVD-BTU mod doesn't fit ball

Auncaughbove17, 2022-03-18:
Hi! I let someone print me the [REVD version](https://github.com/ploopyco/classic-trackball/blob/master/hardware/Mechanicals-BTU-Mod/STL/Ploopy-Top-REVD-BTU.stl) of the BTU mod, 
but the ball doesn't fit. With or without the BTUs in place. The hole seems too small..
Here are some pictures comparing the original top part with the new BTU part. Am I doing something wrong? Should I sand down the new part?

[image]
[image]
[image]
[image]

gbrnt, 2022-03-25:
As mentioned in the reddit comments about this, this is because I missed something when making the REVC/REVD changes to the BTU mod. In your screenshot you can see that the REVD top has a much wider hole, while on the BTU top the curve extends further over the ball. I should be able to fix this.

Also, in your screenshot I don't think that's the REVD BTU top - it doesn't have the extra piece that hides the BTU. That was an earlier version. It's not the cause of your issue as the one you've printed isn't the same one that's in the screenshot.

gbrnt, 2022-03-25:
It looks like I lost track of versions at some point (because my original Rhino files aren't tracked in Git since they're large) and applied the REVD changes to the REVA BTU top instead of the REVD BTU top. There's a bit of a mess for me to sort out here.

gbrnt, 2022-03-28:
This should be fixed in [this commit in my fork of the repo](https://github.com/gbrnt/ploopy-classic/commit/25cbf3994e283b2e497a7e64d7ee26e42d14b42b). Please give it a try and report back, then I can ask ploopyco to merge it!

Auncaughbove17, 2022-03-28:
Ok, Thanks! I will let this file be printed now. I will let you know how it turned out.

Auncaughbove17, 2022-04-04:
> This should be fixed in [this commit in my fork of the repo](https://github.com/gbrnt/ploopy-classic/commit/25cbf3994e283b2e497a7e64d7ee26e42d14b42b). Please give it a try and report back, then I can ask ploopyco to merge it!

Now it fits! So, your changes worked!

closed 2022-04-04 by Auncaughbove17