ploopyco/classic-trackball #32: Purchase electronics only?

JamesNewton, 2022-12-23:
Is there no option to purchase /just/ the electronics? I have printers, I have general hardware, I'd really like to just buy the PCB and maybe the cable? No?

gbrnt, 2022-12-24:
I've seen this asked a few times on Reddit and the answer has been that the printed parts are so cheap in comparison to the rest there's not really any reason *not* to send them.

cyberhuman, 2023-03-22:
They may be cheap but some people want to be sustainable and not generate more garbage than really needed :-(

imme-emosol, 2023-08-25:
Would you buy the option if it was priced the same?

Somewhat related is that I would be interested in offering the extra parts from all recycled plastics.

imme-emosol, 2023-08-26:
Issue can probably be closed
~ https://blog.ploopy.co/why-we-dont-offer-kits-without-3d-printed-parts-52 .

closed 2024-04-08 by ploopyco